cp2004/OctoPrint-GCodeMacros #1: skipping the @?

Charly333, 2021-08-10:
Great Plugin that nearmost solves an old problem of mine. On toolchange there are special commands to be inserted depending on what tool is selected. Now I might add an @T1 in to the gcode to start that related marcro for tool 1 - but if T1 only could trigger that macro it would be much easier to swap slicer. Any chance to archive this?

cp2004, 2021-08-10:
To clarify, and make sure I understand this right, you want the macro in `@t1` to run only while T1 is selected? I'm not sure if it's possible to know what tool is running on the printer from OctoPrint at any given time.

Charly333, 2021-08-10:
I´ve 4 tools. Toolchange is triggered by a T1, T2, T3, T4 command in gcode that´s directly inserted by slicer. Whenever one of these T codes occures in the gcode while being processed by octoprint the corresponding macro should be triggered. Each macro will have different servo commands to bring the current tool down on working height. T1 in goce is trigger for @T1 macro. Hope could make it a bit clearer!?

cp2004, 2021-08-10:
Ok I understand now.

It would technically be very easy to do that from a plugin, but it is already supported by OctoPrint:
[image]

The documentation for the variables injected into these scripts should be on https://docs.octoprint.org/en/master/features/gcode_scripts.html#context but I note it is missing the toolchange scripts. You can access `tool.old` and `tool.new`, to create a script that looks something like this in either box:
```jinja
{% if tool.new == 1 %}
.... do T1 toolchange stuff
{% elif tool.new == 2 %}
.... do T2 toolchange stuff
{% endif %}
```
I will get the missing documentation up on docs.octoprint.org soon.

In addition, I want to keep it locked to starting with the `@` for two reasons - first is performance, otherwise I would slow down the gcode sending which can be fragile on slower devices - with 6x 4 character macros, it is around 4x slower on my Pi 4, increasing with each command and character count. Second is simplicity - the plugin is far easier to use if it is clear what is a macro and what is not.

Charly333, 2021-08-10:
Thank´s for the explanations. That´s the missing link I´ve been stuggeling that way. I´ll check it out the next days.

cp2004, 2021-08-22:
Since there has been no new activity on this issue for a while, I'm going to assume the problem was solved. Feel free to comment or open new issues for unsolved problems.

closed 2021-08-22 by cp2004