cp2004/OctoPrint-GCodeMacros #3: Conflict with Bed Visualizer?

jp-001, 2021-08-30:
Version 1.1.0 breaks Bed Visualizer.
FYI, Bed Visualizer uses a "macro" (@BEDLEVELVISUALIZER) for its operation.
I suspect a conflict.

jneilliii, 2021-08-30:
Linking relevant issue. https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/464

jneilliii, 2021-08-30:
I suspect the issue is [here](https://github.com/cp2004/OctoPrint-GCodeMacros/blob/f233ecccf6713829b6280f64cb0c95028f3980dd/octoprint_gcode_macro/__init__.py#L86) @cp2004. That basically means don't return any commands through the remainder of the callbacks doesn't it. Shouldn't you be returning what comes in to go to the next plugin's hooks....

cp2004, 2021-08-30:
That was a stupid mistake, sorry! Should be fixed with the above commit, and I'll release 1.1.1 to get this out to people.

cp2004, 2021-08-30:
That should be good to go. It may take up to 24 hours for the update notification to arrive, or you can force check to install version 1.1.1. Thanks for bringing this to my attention!

closed 2021-08-30 by cp2004

jp-001, 2021-08-30:
Thank you for acting so fast.