cp2004/OctoPrint-GCodeMacros #7: Jinja2 not accessing other plugins

jessemillar, 2022-01-01:
I'm attempting to port some GCODE into a macro. My GCODE commands use Jinja2 to access values from other plugins. Works fine when I have the same GCODE in the Octoprint "GCODE Scripts" section but when I try to run my macro after defining it in GcodeMacros, it just doesn't run. I don't see any GCodeMacros log files and there's no terminal output when I try to run my macro.

```
M104 S{{ plugins.preheat.tool0|int() }} ;Heat nozzle
M109 S{{ plugins.preheat.tool0|int() }} ;Wait for nozzle
M702 ;Unload filament
```

cp2004, 2022-01-02:
The log file is the main `octoprint.log`. You'll probably see an error rendering the script output.

I haven't designed it to take in the same context as the rest of the gcode scripts, although I can see how that might be useful to have for things like smart preheat. I will look at the feasibility (there may be a simple thing I can import and use) of putting the same context into these scripts.

closed 2022-07-07 by cp2004

jessemillar, 2022-07-19:
Totally understandable. Thanks for looking into it!