cp2004/OctoPrint-GCodeMacros #8: Jinja2 syntax

Ricky-cana, 2022-01-28:
Hi all,

I'm trying to use Jinja2 syntax in the plugin.
I'm actually trying to set a macro saying if the bed temp is > than 27 please wait. I set this code but it's not working. Can you tell me where the mistake is? 
Thanks

Code:
{% for n in range(100) %}
  {% if bed_temperature[0] > 27 %}
    G4 S60
  {% endif %}
{% endfor %}

cp2004, 2022-01-28:
The `bed_temperature` variable is not defined anywhere.

Ricky-cana, 2022-01-28:
Hi @cp2004.

Actually, the bed_temperature is the only part that works.
I'm using a Prusa and the bed_temperature[0] variable is the current temp of the bed.

The problem is that the "for" is not respected by the printer.

cp2004, 2022-01-28:
I think you are confused with how this plugin works. It is run in OctoPrint, and the templates are rendered all in one go when the macro command is called. 

There is no `bed_temperature` variable in the plugin. If it was going to work, your loop would either generate 100 `G4 S60` commands and send them to the printer, or generate none, depending on that variable at the time it was sent. I don't think this is what you are trying to create.

closed 2022-02-16 by cp2004