cp2004/OctoPrint-GCodeMacros #12: [Feature Request

cp2004, 2022-07-14:


napter, 2022-07-18:
I just came here to ask if this was possible / post a feature request! My primary use case is to be able to selectively run parts of the script by sending a parameter. 

Another use case, if the variable can retain its value across macros, would be to set a printer name and then later use that to determine things like print temperature.

CmdrCody51, 2022-08-10:
I have 'parameters' working to an extent.
I use this to decide if I want to do a First_Layer check on my prints.

1. Have GCODE Systems Commands plugin. This uses an 'OCTO#' to call shell scripts from GCode files or the 'terminal'.
2. Have Custom Control Editor plugin. This puts buttons on the Control tab.
3. Have Gcode Macros plugin.

1.) I use OCTO300 to turn my flag 'false' and OCTO301 to turn my flag 'true'. (I have a lot of them)
     The 'flag' resides in '/home/pi/.octoprint/data/gcode_macro/my_Logic.flag'.
      The script for OCTO300 is: sed -i 's/true/false/' /home/pi/.octoprint/data/gcode_macro/my_Logic.flag
      The script for OCTO301 is: sed -i 's/false/true/' /home/pi/.octoprint/data/gcode_macro/my_Logic.flag
      The file itself is: {% set mine.my_State = true %}  {Notice the mine.! More later.)
2.) I define two buttons in Custom Controls: Logic True and Logic False. They just contain the OCTO codes above.
3.) In Gcode Macros I define @First_Layer as:
     {% set mine = namespace(my_State=false) %} {% set message = 'First Layer Check' %} {% include 'Do_Pause.macro' %}
     Now... the namespace allows variables to be passed through jinga2 scripts to if/for loops.
4.) I manually enter the following script to /home/pi/.octoprint/data/gcode_macro/ called "Do_Pause.macro"

     {% include 'my_Logic.flag' %}
     {% if mine.my_State %}
     M400 ; complete all motion
     G60 ; store XYZF (fw)
     G91 ; incremental
     G1 Z1 ; lift Z to clear part
     G90 ; back to absolute
     G1 X2 Y217 ; clear the build area
     M400 ; complete all motion
     OCTO203 ; turn lights on
     {% if message is defined %}
     M117 {{ message }}
     {% endif %}
     M300 P1000 ; beep if you got em
     M1 ; pause using printers pause
     OCTO202 ; turn lights off
     G61 ; return to saved position (fw)
     {% endif %}

G60 and G61 stores and restores X,Y,Z and F in current Marlin firmware.
I can see placing many variables into a 'Settings.macro' and running such things as temperature, feedrate and flowrate modifiers.

jneilliii, 2024-03-31:
Have a perfect use case for this (<https://community.octoprint.org/t/how-to-get-the-extruder-that-is-enabled-in-gcode-script-after-print-job-is-paused/57694/4?u=jneilliii>)and potential idea for making the parameters available in other macros. If you were to use environment variables similar to GCode System Commands plugin, you could then do `template.render(env=os.environ)` couldn't you to extract the data out?